open-ephys-plugins/acquisition-board #4: Check that acquisition is not running before initializing impedance measurement

ChucklesOnGitHub, 2024-12-16:
This could help users that inadvertently try to run the impedance measurement while they are recording, crashing the GUI and losing data.

[image]
[activity_2024-12-16_09-50-07_impedances_kill.log](https://github.com/user-attachments/files/18148094/activity_2024-12-16_09-50-07_impedances_kill.log)

jsiegle, 2024-12-16:
I believe the impedance button auto-disables during acquisition in previous versions of this plugin, so it must be something that changed after switching to the new API. We will definitely fix this before the v1.0 release!

jonnew, 2025-01-07:
@jsiegle I assigned you but feel free to delegate to the most appropriate person.

bparks13, 2025-01-09:
Looking at the activity log, this appears to be for the `OE FPGA Acquisition Board` plugin, not the unified `Acquisition Board` plugin in this repo. I confirmed that in the rhythm-oni-plugin it does not check if acquisition is active before running the impedance measurement.

Testing of this plugin does not allow measuring impedances while acquisition is running. The button is still clickable, but it has no affect as internally the plugin is checking if acquisition is active before running the impedance measurements. It is probably still useful to disable the button during acquisition so there is no confusion about whether or not impedances can be measured, but as of right now this behavior is not reproducible for me.

bparks13, 2025-01-23:
Closing this issue since it seems to not be relevant for this repo. Testing shows that impedance will not run during acquisition.