open-ephys-plugins/acquisition-board #10: [macOS] Universal binaries for liboni and related drivers

anjaldoshi, 2025-01-08:
With the release of GUI v1.0, we’re switching to universal binaries (supports both `arm64` and `x86_64` archs) on macOS. Which means the GUI will run natively on both Intel and Apple Silicon Macs. This means that all the plugins need to be built as universal as well. If a plugin uses any external libraries, those libraries also need to be universal. For OE FPGA acquisition boards, we need to build `liboni.dylib`, `libonidriver_ft600.dylib`, and `libftd3xx.dylib` as universal binaries.

anjaldoshi, 2025-01-08:
Since `liboni` lacks build files for macOS, I can’t build it as a universal library on my own. However, I managed to obtain the `x86_64` and `arm64` binaries separately for `libftd3xx`. This allows me to create a universal binary by merging the two architecture-specific binary files using the `lipo` tool.

@aacuevas can you assist me with this?

aacuevas, 2025-01-08:
The current version of libonidriver_ft600.dylib does not work with the latest mac drivers, but the provided older ones that do not support the arm architecture.
@bparks13, we need to expedite the update to the new ftdi driver architecture so this can be done

aacuevas, 2025-01-20:
@anjaldoshi Can you try with the [osx branch](https://github.com/open-ephys/liboni/tree/osx/api/liboni) of the liboni repo? If it works we will merge that into master.

The makefiles should work now for both linux and osx (using clang and command line)
also they:
1. static link the version of libftd3xx that is on the liboni repo, so we do not have to worry about distributing that file (or about permissions and stuff due to that file)
2. should create a Mach-O dylib file with both architectures, x86_64 and arm64

anjaldoshi, 2025-01-23:
@aacuevas I successfully generated the universal binaries for both `iboni.dylib` and `libonidriver_ft600.dylib` using the [osx branch](https://github.com/open-ephys/liboni/tree/osx). However, I had to make minor modifications to ensure successful compilation on my Apple Silicon Mac. Here are the changes I made: https://github.com/anjaldoshi/liboni/commit/ad9978cd7f5b80b8e89ce13c6eefaf81dbf2ce29

aacuevas, 2025-01-23:
Your compiler seems to be pickier than mine, those didn't register even as warnings on any of the compilers I use, but changing definitions to be _stricter_ never hurts.

Can you please make a PR from your branch against the osx branch of the liboni repo? So we can merge into it and probably soon merge that branch into main.

anjaldoshi, 2025-01-23:
PR is up here https://github.com/open-ephys/liboni/pull/29

closed 2025-01-28 by anjaldoshi